SeeedJP/wio_cellular #21: VSCode+PlatformIOでコンパイラのC++標準が古いとエラーになった？

mnltake, 2024-12-08:
```
In file included from C:\Users\username\.platformio\packages\framework-arduinoadafruitnrf52\cores\nRF5\nordic\nrfx\drivers\include/nrfx_spim.h:38:0,
                 from C:\Users\username\.platformio\packages\framework-arduinoadafruitnrf52\cores\nRF5\nordic\nrfx\drivers\src\nrfx_spim.c:42:
C:\Users\username\.platformio\packages\framework-arduinoadafruitnrf52\cores\nRF5\nordic\nrfx/haly/nrfy_spim.h: In function '__nrfy_internal_spim_events_process':
C:\Users\username\.platformio\packages\framework-arduinoadafruitnrf52\cores\nRF5\nordic\nrfx/haly/nrfy_spim.h:838:9: error: expected expression before '[' token
         [[maybe_unused]] size_t size = stop ? nrf_spim_rx_amount_get(p_reg) : p_xfer->rx_length;
         ^
C:\Users\username\.platformio\packages\framework-arduinoadafruitnrf52\cores\nRF5\nordic\nrfx/haly/nrfy_spim.h:838:11: error: 'maybe_unused' undeclared (first use in this function); did you mean '__unused'?
         [[maybe_unused]] size_t size = stop ? nrf_spim_rx_amount_get(p_reg) : p_xfer->rx_length;
           ^~~~~~~~~~~~
           __unused
C:\Users\username\.platformio\packages\framework-arduinoadafruitnrf52\cores\nRF5\nordic\nrfx/haly/nrfy_spim.h:838:11: note: each undeclared identifier is reported only once for each function it appears in
C:\Users\username\.platformio\packages\framework-arduinoadafruitnrf52\cores\nRF5\nordic\nrfx/haly/nrfy_spim.h:846:9: error: expected expression before '[' token
         [[maybe_unused]] size_t size = stop ? nrf_spim_rx_amount_get(p_reg) : p_xfer->rx_length;
         ^
C:\Users\username\.platformio\packages\framework-arduinoadafruitnrf52\cores\nRF5\nordic\nrfx/haly/nrfy_spim.h:833:10: warning: unused variable 'stop' [-Wunused-variable]
     bool stop = __nrfy_internal_spim_event_handle(p_reg, mask, NRF_SPIM_EVENT_STOPPED, &evt_mask);
          ^~~~
*** [.pio\build\seeed_wio_bg770a\FrameworkArduino\nordic\nrfx\drivers\src\nrfx_spim.c.o] Error 1
```

上記のエラーが出ましたがChatGPTに聞いた所
> このエラーメッセージは、コードに使用されている[[maybe_unused]]属性が適切に認識されないことに起因しています。この問題は、コンパイラがC++17以降を使用していない場合や、コードがC言語としてコンパイルされている場合に発生します。

とあり色々試した所結論だけ言うと

```
 pio pkg update
```
で解決しました　お騒がせしました

matsujirushi, 2024-12-08:
情報ありがとうございます！

状況から、コンパイルオプションが更新されていなかったように思われます。
コンパイル時の「PACKAGES」を知りたいのですが、残っていませんか？

mnltake, 2024-12-09:
DesktopPCでビルド成功したのが
```
PACKAGES:
 - framework-arduinoadafruitnrf52 @ 1.10300.0+sha.48055dd
 - framework-cmsis @ 2.50700.210515 (5.7.0)
 - tool-adafruit-nrfutil @ 1.503.0 (5.3)
 - tool-sreccat @ 1.164.0 (1.64)
 - toolchain-gccarmnoneeabi @ 1.120301.0 (12.3.1)
 ````

昨日持っていったタブレットのでまだエラーが出てるのが
```
PACKAGES:
 - framework-arduinoadafruitnrf52 @ 1.10300.0+sha.48055dd
 - framework-cmsis @ 2.50700.210515 (5.7.0)
 - tool-adafruit-nrfutil @ 1.503.0 (5.3)
 - tool-sreccat @ 1.164.0 (1.64)
 - toolchain-gccarmnoneeabi @ 1.70201.0（7.2.1）
```

pio run -v 
で細かく出すと
-std=gnu++11 のとこが　-std=gnu++20　に変わってます

mnltake, 2024-12-09:
commit 3b108ef9f314b2f2c00bdcb65cb091eb220f54ab (grafted, HEAD -> main, origin/main, origin/HEAD)
Author: MATSUOKA Takashi <matsujirushi@live.jp>
Date:   Tue Aug 20 16:17:04 2024 +0900

    Add board Wio BG770A

matsujirushi, 2024-12-09:
ありがとうございます！
送っていただいた情報で、こちらで再現テストできると思うので、確認します。

matsujirushi, 2024-12-13:
クリーン環境からAdafruit Feather nRF52840 -> Wio BG770Aへの切り替えは正常。

# Adafruit Feather nRF52840の開発環境をセットアップ

1. PlatformIOのplatformとframeworkを削除

2. Platformをインストール

    * Nordic nRF52 10.7.0

3. arduino-blinkをインポート

4. platformio.iniを変更

    ```
    [env:adafruit_feather_nrf52840]
    platform = nordicnrf52
    framework = arduino
    board = adafruit_feather_nrf52840
    ```

5. ビルド

    ```
    CONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf52/adafruit_feather_nrf52840.html
    PLATFORM: Nordic nRF52 (10.7.0) > Adafruit Feather nRF52840 Express
    HARDWARE: NRF52840 64MHz, 243KB RAM, 796KB Flash
    DEBUG: Current (blackmagic) External (blackmagic, cmsis-dap, jlink, stlink)
    PACKAGES: 
     - framework-arduinoadafruitnrf52 @ 1.10601.0 (1.6.1) 
     - framework-cmsis @ 2.50700.210515 (5.7.0) 
     - tool-adafruit-nrfutil @ 1.503.0 (5.3) 
     - tool-sreccat @ 1.164.0 (1.64) 
     - toolchain-gccarmnoneeabi @ 1.70201.0 (7.2.1)
    ```

# Wio BG770Aのbasic/blinkを開く

```
Resolving seeed_wio_bg770a dependencies...
Platform Manager: Installing git+https://github.com/SeeedJP/platform-nordicnrf52
git version 2.40.1.windows.1
Cloning into 'C:\Users\takashi\.platformio\.cache\tmp\pkg-installing-quzsjash'...

Submodule 'builder/frameworks/arduino' (https://github.com/SeeedJP/builder-framework-arduino-nrf5.git) registered for path 'builder/frameworks/arduino'

Cloning into 'C:/Users/takashi/.platformio/.cache/tmp/pkg-installing-quzsjash/builder/frameworks/arduino'...

Submodule path 'builder/frameworks/arduino': checked out '7158d3e1a873b429fcf0eb15f79b2163d8ca0a3a'
Submodule 'mbed-core' (https://github.com/platformio/builder-framework-arduino-core-mbed.git) registered for path 'builder/frameworks/arduino/mbed-core'

Cloning into 'C:/Users/takashi/.platformio/.cache/tmp/pkg-installing-quzsjash/builder/frameworks/arduino/mbed-core'...

Submodule path 'builder/frameworks/arduino/mbed-core': checked out '66ff9c5ecf64e0ce6360cbf2d8769228bb1294b4'
Platform Manager: nordicnrf52@10.1.0+sha.8f55ff2 has been installed!
Tool Manager: Installing platformio/toolchain-gccarmnoneeabi @ ~1.120301.0
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Tool Manager: toolchain-gccarmnoneeabi@1.120301.0 has been installed!
Tool Manager: Installing git+https://github.com/SeeedJP/Adafruit_nRF52_Arduino.git
git version 2.40.1.windows.1
Cloning into 'C:\Users\takashi\.platformio\.cache\tmp\pkg-installing-xzvfhsre'...
...
Updating files: 100% (1210/1210), done.

Submodule 'libraries/Adafruit_TinyUSB_Arduino' (https://github.com/adafruit/Adafruit_TinyUSB_Arduino.git) registered for path 'libraries/Adafruit_TinyUSB_Arduino'
Submodule 'libraries/Adafruit_nRFCrypto' (https://github.com/adafruit/Adafruit_nRFCrypto.git) registered for path 'libraries/Adafruit_nRFCrypto'

Cloning into 'C:/Users/takashi/.platformio/.cache/tmp/pkg-installing-xzvfhsre/libraries/Adafruit_TinyUSB_Arduino'...

Cloning into 'C:/Users/takashi/.platformio/.cache/tmp/pkg-installing-xzvfhsre/libraries/Adafruit_nRFCrypto'...

Submodule path 'libraries/Adafruit_TinyUSB_Arduino': checked out 'e58ed11de15e2cfbd71f1a79cf84a2372ba27e08'
Submodule path 'libraries/Adafruit_nRFCrypto': checked out '7444c2e9472b8731cbcb76be5216df04680c5529'
Tool Manager: framework-arduinoadafruitnrf52@1.10300.0+sha.48055dd has been installed!
Updating metadata for the vscode IDE...
Project has been successfully updated!
```

# basic/blinkをビルド

ビルド成功。

```
CONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf52/seeed_wio_bg770a.html
PLATFORM: Nordic nRF52 (10.1.0+sha.8f55ff2) > Wio BG770A
HARDWARE: NRF52840 64MHz, 232KB RAM, 780KB Flash
DEBUG: Current (blackmagic) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES: 
 - framework-arduinoadafruitnrf52 @ 1.10300.0+sha.48055dd 
 - framework-cmsis @ 2.50700.210515 (5.7.0) 
 - tool-adafruit-nrfutil @ 1.503.0 (5.3) 
 - tool-sreccat @ 1.164.0 (1.64) 
 - toolchain-gccarmnoneeabi @ 1.120301.0 (12.3.1)
 ```

matsujirushi, 2024-12-13:
platform-nordicnrf52がインストールされた後に、GitHubを更新したとき。

1. GitHub更新後にbasic/blinkを開く

    ```
    Resolving seeed_wio_bg770a dependencies...
    Already up-to-date.
    Updating metadata for the vscode IDE...
    Project has been successfully updated!
    ```

2. ビルド

    ビルドがエラーになる。
    platform-nordicnrf52が更新されていない。
    
    ```
    CONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf52/seeed_wio_bg770a.html
    PLATFORM: Nordic nRF52 (10.1.0+sha.3b108ef) > Wio BG770A
    HARDWARE: NRF52840 64MHz, 232KB RAM, 780KB Flash
    DEBUG: Current (blackmagic) External (blackmagic, cmsis-dap, jlink, stlink)
    PACKAGES: 
     - framework-arduinoadafruitnrf52 @ 1.10300.0+sha.48055dd 
     - framework-cmsis @ 2.50700.210515 (5.7.0) 
     - tool-adafruit-nrfutil @ 1.503.0 (5.3) 
     - tool-sreccat @ 1.164.0 (1.64) 
     - toolchain-gccarmnoneeabi @ 1.70201.0 (7.2.1)
     ```

    どうやら、PlatformIOのplatformは自動更新されない模様。 
    platformのUpdatesに更新対象が表示される。
    
    [image]
    [image]
    
3. GUIでplatform-nordicnrf52をアップデート

    WARNING表示が出たが、アップデート完了した模様。
    
    [image]
    
4. basic/blinkをビルド

    ビルド成功。

    ```
    CONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf52/seeed_wio_bg770a.html
    PLATFORM: Nordic nRF52 (10.1.0+sha.8f55ff2) > Wio BG770A
    HARDWARE: NRF52840 64MHz, 232KB RAM, 780KB Flash
    DEBUG: Current (blackmagic) External (blackmagic, cmsis-dap, jlink, stlink)
    PACKAGES:
     - framework-arduinoadafruitnrf52 @ 1.10300.0+sha.48055dd
     - framework-cmsis @ 2.50700.210515 (5.7.0)
     - tool-adafruit-nrfutil @ 1.503.0 (5.3)
     - tool-sreccat @ 1.164.0 (1.64)
     - toolchain-gccarmnoneeabi @ 1.120301.0 (12.3.1)
    ```

mnltake, 2024-12-13:
なるほど
前にArduinoNano33BLEを使ったときにplatform-nordicnrf52を入れたはずです
（updateのお知らせはあったかもしれないけど無視してた）
libraryだけじゃなくplatformのバージョンも気にしなきゃいけないんですね

matsujirushi, 2024-12-13:
CLIによる確認方法。

既に最新版のとき:
```
$ pio pkg outdated
Checking  [####################################]  100%
Everything is up-to-date!
```

最新版になっていないとき:
```
$ pio pkg outdated
Checking  [####################################]  100%
...
Package      Current             Wanted    Latest              Type      Environments
-----------  ------------------  --------  ------------------  --------  ----------------
nordicnrf52  10.1.0+sha.3b108ef            10.1.0+sha.8f55ff2  Platform  seeed_wio_bg770a
```

最新版に更新:
```
pio pkg update
Resolving seeed_wio_bg770a dependencies...
Platform Manager: Updating nordicnrf52 @ 10.1.0+sha.3b108ef
git version 2.40.1.windows.1
remote: Enumerating objects: 23, done.
remote: Counting objects: 100% (23/23), done.
remote: Compressing objects: 100% (9/9), done.
remote: Total 13 (delta 9), reused 8 (delta 4), pack-reused 0 (from 0)
Unpacking objects: 100% (13/13), 1.22 KiB | 27.00 KiB/s, done.
From https://github.com/matsujirushi/platform-nordicnrf52
   3b108ef..8f55ff2  main       -> origin/main
Fetching submodule builder/frameworks/arduino
Fetching submodule builder/frameworks/arduino/mbed-core
From https://github.com/platformio/builder-framework-arduino-nrf5
 * branch            7158d3e1a873b429fcf0eb15f79b2163d8ca0a3a -> FETCH_HEAD
Fetching submodule builder/frameworks/arduino/mbed-core
Updating 3b108ef..8f55ff2
Fast-forward
 .gitmodules                   | 2 +-
 boards/seeed_wio_bg770a.json  | 2 +-
 builder/frameworks/arduino    | 2 +-
 builder/frameworks/arduino.py | 2 +-
 builder/main.py               | 2 +-
 platform.json                 | 6 +-----
 platform.py                   | 2 +-
 7 files changed, 7 insertions(+), 11 deletions(-)
From https://github.com/platformio/builder-framework-arduino-nrf5
 * branch            7158d3e1a873b429fcf0eb15f79b2163d8ca0a3a -> FETCH_HEAD
Submodule path 'builder/frameworks/arduino': checked out '7158d3e1a873b429fcf0eb15f79b2163d8ca0a3a'
Submodule path 'builder/frameworks/arduino/mbed-core': checked out '66ff9c5ecf64e0ce6360cbf2d8769228bb1294b4'
Already up-to-date.
```

matsujirushi, 2024-12-13:
# 原因

wio_cellularがPlatformIOにうまく対応できていない開発環境（ #11 ）でビルドしている。
自動的に最新の開発環境にアップグレードされないため、何度試してもエラーが解消されない。

# 解決方法

開発環境を最新にアップデートする。

PlatformIO Core CLI:
```
$ pio pkg update
```

closed 2024-12-13 by matsujirushi