SeeedJP/wio_cellular #25: soracom-uptime-lp.inoのWioCellular.powerOn()で止まる

susami-lab, 2024-12-15:
現状、soracom-uptime-lp.inoを実行すると、数10分から数時間で止まってしまいます。
どのような要因が考えられるでしょうか。

・止まる場所は、下記です。
if (measure(JsonDoc)) {
      ABORT_IF_FAILED(WioCellular.powerOn(POWER_ON_TIMEOUT));
      send(JsonDoc);
}

・シリアルプリントは下記です。
14:22:13.980 -> URC> PSM POWER DOWN
14:26:35.270 -> ### Measuring
14:26:35.270 -> ### Completed
14:26:37.025 -> URC> RDY
14:26:37.025 -> URC> +CFUN: 1
14:26:37.025 -> URC> +CPIN: READY
14:26:37.025 -> URC> +QUSIM: 1
14:26:37.025 -> URC> +QIND: SMS DONE
14:26:38.030 -> URC> APP RDY
（中略）
14:26:40.731 -> ### Completed
14:27:12.210 -> URC> +QPSMTIMER: 360,2
14:27:14.217 -> URC> +CEREG: 4
14:27:14.826 -> URC> PSM POWER DOWN
14:31:35.331 -> ### Measuring
14:31:35.331 -> ### Completed
この後、
URC> RDY
が返ってきません。

matsujirushi, 2024-12-15:
ご連絡ありがとうございます。
まずは、こちらで事象が再現するかを確認します。

matsujirushi, 2024-12-16:
# 1台目

6時間動かしたが、止まらず。

[image]

[image]

matsujirushi, 2024-12-16:
メモ。
電源ONに関して。

* モジュール仕様 ... PWRKEYを500-1000ms、LOW。
* コード ... 502ms。https://github.com/SeeedJP/wio_cellular/blob/42ac1565cecaf71aac6fa332a83c03f16de18508/src/module/bg770a/Bg770aInterface.hpp#L184
* 実測 ... 501.96676ms

matsujirushi, 2024-12-16:
# 2台目

すぐに再現した。

```
[2024-12-16 15:08:06.188] ### Completed
[2024-12-16 15:08:37.437] URC> +QPSMTIMER: 360,2
[2024-12-16 15:08:39.429] URC> +CEREG: 4
[2024-12-16 15:08:40.041] URC> PSM POWER DOWN
[2024-12-16 15:13:00.697] ### Measuring
[2024-12-16 15:13:00.697] ### Completed
```
[uptime-lp.log](https://github.com/user-attachments/files/18145722/uptime-lp.log)

[image]

* USER LED ... 点灯
* STATUS ... 点灯
* NET ... 点灯（←ん？）

[image]

susami-lab, 2024-12-16:
```
if (measure(JsonDoc)) {
      auto result = WioCellular.powerOn(POWER_ON_TIMEOUT);
      if (result != WioCellularResult::Ok) {
        Serial.println("Power-on failed.");
        // エラー発生時もプログラムを続行
      } else {
        send(JsonDoc);  // 電源オンが成功した場合のみ送信
      }
}
```
このようにコードを改変し、試験をした結果を添付します。
前半と後半で試験場所が異なるのですが、明らかに差があります。
WioCellular.powerOn()と関係あるかはわかりませんが、
前半は、iPhoneのアンテナ本数1本と、電波環境が悪いです。
電波と関係があるものでしょうか。一つの可能性として情報共有します。
[image]

matsujirushi, 2024-12-16:
@susami-lab 

今日も2台目でテストしていますが、なかなか再現しない状況です。

以下、分かる範囲で結構ですので教えていただけませんか？

1. 止まったときのLEDは？

    わたしの手元で1度だけ止まったときは、以下でした。

    * USER LED ... 点灯
    * STATUS LED ... 点灯
    * NET LED ... 点灯

2. 止まったときはいつも同じLEDか？

susami-lab, 2024-12-17:
試験開始20分で再現しました。
止まったときのLEDは下記です。

- USER LED ... 点灯
- STATUS LED ... 点灯
- NET LED ... 点灯

susami-lab, 2024-12-17:
開始15分で2回目再現しました。
停止3分後のLED状態は下記です。

- USER LED ... 点灯
- STATUS LED ... 点灯
- NET LED ... 点灯

停止直後のLED状態は確認できておりません。

susami-lab, 2024-12-17:
試験開始50分で3回目再現しました。
停止6.5時間後のLED状態は下記です。
NET LEDの状態が、1,2回目と異なります。

USER LED ... 点灯
STATUS LED ... 点灯
NET LED ... 点灯200msec/消灯1800msec

matsujirushi, 2024-12-18:
@susami-lab 
ご確認ありがとうございます。

1,2回目はセルラーモジュールが止まってしまっているように見えます。（NET LEDが点灯）
3回目はセルラーモジュールがNETに接続できていないようです。（NET LEDが短く点灯）

こちらで電波状況を悪くして確認してみます。

（アンテナ外してもNET接続できてしまう環境なので、、、会社からシールドボックス持ってこようと思っています。）

susami-lab, 2024-12-19:
4回目再現しました。

停止5時間後のLED状態は下記です。
USER LED ... 点灯
STATUS LED ... 点灯
NET LED ... 点灯

そのまま放置して停止12時間後のLED状態は下記です。NET LEDの状態が変わっています。
USER LED ... 点灯
STATUS LED ... 点灯
NET LED ... 点灯1800msec/消灯200msec

susami-lab, 2024-12-19:
あっ3回目のNET LEDの点灯消灯を逆に書いたかもしれません。。少なくとも、4回目のNET LED状態（点灯が長く、消灯が短い）は間違いないです。

matsujirushi, 2024-12-20:
いままでの情報ならびにシールドボックスでセルラー通信できない状況を作ってテストした結果から、2つの事象があるようです。

1. セルラーネットワークに接続していないとPSMに遷移しない

    PSMに遷移しないため、セルラーモジュールがOFFせず消費電流が下がらない。
    （コードが、セルラーネットワークに接続できない状態を考慮していない。）

2. セルラーモジュールのNET LEDが点灯

    セルラーモジュールが不明な挙動。
    （データシートにNET LED点灯が書かれていない。）

まずは、1.を対処します。

matsujirushi, 2025-01-31:
1.について、サンプルコードsoracom/soracom-uptime-psm.inoで対処しました。[v0.3.2](https://github.com/SeeedJP/wio_cellular/releases/tag/0.3.2)でリリース。

https://github.com/SeeedJP/wio_cellular/blob/main/examples/soracom/soracom-uptime-psm/soracom-uptime-psm.ino

matsujirushi, 2025-02-25:
@susami-lab 
#38 は事象が発生しない＆再現方法がわからない状況です。
そのため、本件クローズします。

事象が再発するようでしたら、再オープン願います。

closed 2025-02-25 by matsujirushi