SeeedJP/wio_cellular #48: I2Cのピン設定

PikaPikaLight, 2025-05-14:
Wio BG770AとWio Extension – RTCをI2Cでつないで動作させたところ不安定な動作でした。  
オシロスコープでI2Cの波形を確認するとLow時の電圧が1.2Vで信号が落ち切っていませんでした。  
ボードのライブラリ中のWire_nRF52.cppでGPIO_PIN_CNF_DRIVE_S0D1をGPIO_PIN_CNF_DRIVE_H0D1に変更するとHigh driveになって波形が改善しました。  
driveの設定を関数で変更できるようにしていただけるとありがたいです。  
  
またWire_nRF52.cppをみて気づきましたがI2CマスターモードではSDA,SCLが内部プルアップの設定になっています。実際にボード上のプルアップ抵抗をすべて外して確認したところ、I2C信号が出ていたので内部プルアップされているのは間違いないです。できればこれも関数で内部プルアップのOn/Offを変えられるようにしていただけるとありがたいです。

matsujirushi, 2025-05-15:
メモ。

### Wio BG770A + Wio Extension - RTCのI2C波形

* （ご指摘のとおり）SDA/SCLとも、1.2V程度。

[image]

[image]

### I2Cのプルアップ抵抗

* Wio Extension - RTC ... 1K
* [Grove - ADXL345 - 3-Axis Digital Accelerometer(±16g)](https://www.seeedstudio.com/Grove-3-Axis-Digital-Accelerometer-16g.html) ... 4.7K ←Wio BG770AのI2Cテストに使用しているGroveモジュール

### BSPのフォーク元

* GPIO_PIN_CNF_DRIVE_S0D1 & GPIO_PIN_CNF_PULL_Pullup

https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/aa0f46ce6400c18e25f59b11c7c24cd7ef807258/libraries/Wire/Wire_nRF52.cpp#L54-L64

### nRF52840のスペック

* S0 ... Max. 0.4V, Typ. 2mA
* H0 ... Max. 0.4V, Typ. 10mA
* プルアップ抵抗 ... Typ. 13K

### Nordic DevZone情報

* [What are the nRF52840 output drive modes mean](https://devzone.nordicsemi.com/f/nordic-q-a/94697/what-are-the-nrf52840-output-drive-modes-mean)
* [[NRF5340] Can TWIM (I2C) Pin drive strength be changes to H0D1?](https://devzone.nordicsemi.com/f/nordic-q-a/74310/nrf5340-can-twim-i2c-pin-drive-strength-be-changes-to-h0d1)
* [i2c drive strength unsolved after applying forum's method](https://devzone.nordicsemi.com/f/nordic-q-a/64109/i2c-drive-strength-unsolved-after-applying-forum-s-method)

matsujirushi, 2025-05-15:
メモ。
ハイドライブ、プルアップ無しの場合。

https://github.com/SeeedJP/Adafruit_nRF52_Arduino/commit/ea496a4fb2ca05a9218bc2b85ed4a70ec7550f0f

### Wio BG770A + Wio Extenson - RTCのI2C波形

[image]

[image]

### Wio BG770A + ADXL345のI2C波形

[image]

[image]

matsujirushi, 2025-05-15:
@PikaPikaLight 
ご指摘ありがとうございます！

I2Cのデフォルトを、

* スタンダードドライブ
* プルアップ抵抗あり

から、

* ハイドライブ
* プルアップ抵抗なし

に変更しようと思います。

matsujirushi, 2025-05-15:
@PikaPikaLight 
リリース完了しました。
https://github.com/SeeedJP/Adafruit_nRF52_Arduino/releases/tag/1.4.5

どうもありがとうございました。

closed 2025-05-15 by matsujirushi