SeeedJP/wio_cellular #53: 電源電圧を測定できるようにしてほしい

matsujirushi, 2025-05-26:
バッテリー電圧低下の検知のような、電源電圧の監視をしたい。
VSYSをnRF52のADコンバーターで測定できるように機能追加してほしい。

### 参考リンク

* [Wio BG770Aの研究9：自分自身の電源電圧を計測する](https://note.com/susa_lab/n/nf74329b4b07b?magazine_key=m2207f165ba87)

> 自作IoTデバイスにおいて、IoTデバイスへの給電方法が商用電源であれば、電源電圧の計測は不要かもしれませんが、電池やバッテリー等であれば、電源電圧を把握は重要です。IoTデバイスに接続された各種センサーで取得したデータと同時に、自分自身の電源電圧をクラウドに送ることで電源電圧の把握が可能となり、死活監視や適切な電池交換時期の把握等が可能となります。

matsujirushi, 2025-05-29:
分圧抵抗をCJ3407とDTC043でON/OFFした場合の消費電流。

* 分圧抵抗ON
  * 平均 643.68[uA]
* 分圧抵抗OFF
  * 平均 95.98[uA]

[image]

[image]

susami-lab, 2025-06-03:
なるほど。
ADCの精度を高める手段として、
分解能、サンプリング時間、オーバーサンプリングを調整できるのですね。
ありがとうございます。

```
analogReadResolution(14); // ADC分解能を14ビットに設定
analogSampleTime(40); // ADCサンプリング時間を40µsに設定
analogOversampling(256); // 1回のアナログ読み取りにつき256個のサンプルを取り、それらを平均して1つの結果を出力
```

matsujirushi, 2026-10-22:
https://github.com/SeeedJP/wio_cellular/issues/54#issuecomment-6332806168

closed 2026-10-22 by matsujirushi